obmaz/smartthings_edge_driver #3: new button

niquini, 2021-12-21:
it would be possible to create a driver for a button from Zemismart (Tuya) that has 6 keys with 3 actions each.

The data is

zigbeeNodeType: SLEEPY_END_DEVICE
application: 44
endpointId: 01
manufacturer: _TZE200_zqtiam4u
model: TS0601

01 0104 0051 01 04 0000 0004 0005 EF00 02 0019 000A
--

device link
https://s.click.aliexpress.com/e/_A5FXn3

averagebald, 2021-12-23:
Need some help on adding the below as well. Its a 4 button wireless zigbee.

manufacturer: _TZ3000_ee8nrt2l
model: TS0044

obmaz, 2021-12-23:
> Need some help on adding the below as well. Its a 4 button wireless zigbee.
> 
> manufacturer: _TZ3000_ee8nrt2l model: TS0044

I added the FP and it is applied within 24hour. 
or you can use it now with reinstalling (remove and install driver)
Let me know whether it work or not

obmaz, 2021-12-23:
> it would be possible to create a driver for a button from Zemismart (Tuya) that has 6 keys with 3 actions each.
>
> The data is
>
> zigbeeNodeType: SLEEPY_END_DEVICE application: 44 endpointId: 01 [...]

I researched TS0601, and EF00 is a custom cluster used by tuya. So it seems difficult to implement without an actual device.

niquini, 2021-12-23:
> I researched TS0601, and EF00 is a custom cluster used by tuya. So it seems difficult to implement without an actual device.

I understand. If it might be useful, I'll send you the DHT that I use on it

DHT
https://drive.google.com/file/d/1WaoM80xPi2TMsf-Z-itKLr2p7VKFZ5xh/view?usp=sharing

obmaz, 2021-12-23:
> 

OKay.

I started imagining coding, and if this works for a bit, I might be able to implement it through testing, although it may take some time.

I already updated the code (need to reinstall) and please test below
Code:  Mapped the EF00 cluster and 0x01 command to handler. 

1. Click the button1 -> It will trigger click, double click and hold at the same time
2. double Click the button1 -> It will trigger click, double click and hold at the same time
3. Hold the button1 -> It will trigger click, double click and hold at the same time

niquini, 2021-12-23:
> > 
> 
> OK.
> 
> Comecei a imaginar a codificação e, se funcionar por um tempo, talvez consiga implementá-la por meio de testes, embora possa levar algum tempo.
> 
> Já atualizei o código (preciso reinstalar) e teste abaixo Código: Mapeado o cluster EF00 e o comando 0x01 para o manipulador. Ver: 2021-12-23T14: 12: 35.59249
> 
> 1. Clique no botão1 -> Ele acionará o clique, clique duas vezes e segure ao mesmo tempo
> 2. Clique duas vezes no botão1 -> Ele irá acionar o clique, clique duas vezes e segure ao mesmo tempo
> 3. Segure o botão1 -> Ele irá acionar o clique, clique duas vezes e segure ao mesmo tempo

we're walking... only 4 buttons appeared and there are 6. Another thing... when I click it doesn't always give the same action... sometimes there's a click, or two, or keep pressed.... but at least you've already seen the button... the click operation remains to be aligned

obmaz, 2021-12-23:
Please test below

Code: Added 6 button UI / Added click type / All butons mapped to button1
Ver: 2021-12-23T16:19:59.266867

1. Click the button1 -> It will only trigger click
2. double Click the button1 -> It will only trigger double click
3. Hold the button1 -> It will only trigger hold

niquini, 2021-12-23:
> Por favor, teste abaixo
> 
> Código: 6 botões de IU adicionados / Tipo de clique adicionado / Todos os botões mapeados para o botão 1 Ver: 2021-12-23T16: 19: 59.266867
> 
> 1. Clique no botão1 -> Só irá disparar o clique
> 2. Clique duas vezes no botão1 -> Só vai disparar um clique duplo
> 3. Segure o botão 1 -> Isso só vai disparar a espera

6 buttons appeared but none of them have action

obmaz, 2021-12-24:
Please test again

Code: Added click type / All butons mapped to button1
Ver: 2021-12-24T08:19:54.225382

Click the button1 -> It will only trigger click
double Click the button1 -> It will only trigger double click
Hold the button1 -> It will only trigger hold

niquini, 2021-12-24:
Hello my friend, how are you? I removed and installed the driver again and it still doesn't work

obmaz, 2021-12-27:
I corrected the part that I thought would be a problem, but it is a pity that there is no response.

Could you share the log generated using smartthings cli?
 - https://github.com/SmartThingsCommunity/smartthings-cli/releases

obmaz, 2021-12-29:
I upload new version (2021-12-29T11:44:25.982138)
Please test it and let me know

niquini, 2021-12-29:
can we close here and move on to the community?

https://community.smartthings.com/t/edge-driver-for-zemismart-6-button-remote/236849

obmaz, 2021-12-29:
This issue will be managed on community

closed 2021-12-29 by obmaz