obmaz/smartthings_edge_driver #4: Need some help on adding the below as well. Its a 4 button wireless zigbee.

obmaz, 2021-12-23:
Need some help on adding the below as well. Its a 4 button wireless zigbee.

manufacturer: _TZ3000_ee8nrt2l
model: TS0044

_Originally posted by @averagebald in https://github.com/obmaz/smartthings_edge_driver/issues/3#issuecomment-1000158109_

obmaz, 2021-12-23:
Fixed

averagebald, 2021-12-24:
It worked! i had to restart my hub though. All buttons and presses works. Thanks for that. However, seems like the battery status is not getting reported. it shows 0% only. 
[image]

obmaz, 2021-12-24:
Good.
It need some time to get battery status if the device use standard protocol.
If not, the battery status might show incorrect value including 0%

closed 2021-12-24 by obmaz

averagebald, 2021-12-24:
No worries. Important thing is it works. if suddenly it stops working, one thing I would try is replace the battery. Thanks! 

btw, i am opening another issue for my Zemismart zigbee 4-Button. I hope that is fine. Thanks again.