obmaz/smartthings_edge_driver #6: Zigbee-Tuya-Button - _TZ3000_xabckq1v

averagebald, 2021-12-24:
application: 41
endpointId: 01
manufacturer: _TZ3000_xabckq1v
model: TS004F
zigbeeNodeType: SLEEPY_END_DEVICE

i see this added in the fingerprint but somehow it's not working. one thing i notice is its being detected as "Hejhome Smart 4 Button" when added. None of the buttons works.
[image]

obmaz, 2021-12-27:
Yes, right. It might not work currently.
TS004F signal is not the same as TS0044.

I've ordered TS004F from aliexpress for dev.
I might start to implement it next year due to too slow delivery from ali.

I will keep this issue open and add comments when development is complete.

averagebald, 2022-03-24:
Hi mate, 
Just want to follow up on this. Did you manage to get the TS004F? Thanks.

obmaz, 2022-03-24:
Hi,
I stopped developing for the TS004F. The below thread below might be helpful.
- https://community.smartthings.com/t/st-edge-zigbee-driver-for-moes-4-button-scene-switch-ts004f/237808/1

closed 2022-03-24 by obmaz

averagebald, 2022-03-25:
no worries. thanks!