obmaz/smartthings_edge_driver #15: 2min off issue log

7destiny, 2022-06-16:
? Select a driver. 10
2022-06-16T12:07:59.346850483+00:00 DEBUG zigbee-tuya-switch  driver device thread event handled
2022-06-16T12:08:29.347068498+00:00 DEBUG zigbee-tuya-switch  driver device thread event handled
2022-06-16T12:08:37.068611835+00:00 TRACE zigbee-tuya-switch  Received event with handler capability
2022-06-16T12:08:37.086843835+00:00 INFO zigbee-tuya-switch  <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)> received command: {"args":{},"capability":"switch","command":"on","component":"main","positional_args":{}}
2022-06-16T12:08:37.094483835+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.100090835+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.106597501+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.112087501+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.118275168+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.123878835+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.129409168+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.134803835+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.140280168+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.146249168+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.151639835+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.157037168+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.162514168+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.167942835+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.173876168+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.179296168+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.184657835+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.190261168+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.195513501+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.201548835+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.206790168+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.212223835+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.217796835+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.223448501+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.229384835+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch : false
2022-06-16T12:08:37.234793835+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.240321168+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.245706168+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.251107835+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.257292501+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.262718168+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.268404501+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.274062168+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.279972835+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.286052168+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.291453501+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.296857501+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.302224168+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.307626168+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.313545501+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
2022-06-16T12:08:37.319201168+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)>
[... cut ...]
2022-06-16T12:10:53.836419124+00:00 INFO zigbee-tuya-switch  <ZigbeeDevice: 4a111927-51b7-4641-a8b9-85dcc7cbb514 [0x969C] (Lonsonho  Switch 1)> emitting event: {"attribute_id":"switch","capability_id":"switch","component_id":"main","state":{"value":"off"}}
2022-06-16T12:10:53.853526124+00:00 DEBUG zigbee-tuya-switch  Lonsonho  Switch 1 device thread event handled
2022-06-16T12:10:59.376971793+00:00 DEBUG zigbee-tuya-switch  driver device thread event handled
2022-06-16T12:11:29.446815474+00:00 DEBUG zigbee-tuya-switch  driver device thread event handled
2022-06-16T12:11:59.429141155+00:00 DEBUG zigbee-tuya-switch  driver device thread event handled
listening for logs... -

obmaz, 2022-06-17:
FIxed and Deplyed (2022-06-17T05:48:18.992667022)

7destiny, 2022-06-17:
│ 10 │ de2773fd-2901-4254-9009-276f1ca90350 │ zigbee-tuya-switch             │
└────┴──────────────────────────────────────┴────────────────────────────────┘
? Select a driver. 10
2022-06-17T10:52:14.372991115+00:00 DEBUG zigbee-tuya-switch  driver device thread event handled
2022-06-17T10:52:41.567328784+00:00 TRACE zigbee-tuya-switch  Received event with handler capability
2022-06-17T10:52:41.732948784+00:00 INFO zigbee-tuya-switch  <ZigbeeDevice: 8a1f9738-d7b7-42b5-bf61-c9cd7e9a1f76 [0xF8A1] (Lonsonho  Switch 1)> received command: {"args":{},"capability":"switch","command":"on","component":"main","positional_args":{}}
2022-06-17T10:52:41.844762785+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 8a1f9738-d7b7-42b5-bf61-c9cd7e9a1f76 [0xF8A1] (Lonsonho  Switch 1)>
2022-06-17T10:52:41.859724451+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 8a1f9738-d7b7-42b5-bf61-c9cd7e9a1f76 [0xF8A1] (Lonsonho  Switch 1)>
2022-06-17T10:52:41.865079118+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 8a1f9738-d7b7-42b5-bf61-c9cd7e9a1f76 [0xF8A1] (Lonsonho  Switch 1)>
2022-06-17T10:52:41.877146118+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 8a1f9738-d7b7-42b5-bf61-c9cd7e9a1f76 [0xF8A1] (Lonsonho  Switch 1)>
2022-06-17T10:52:41.892318118+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 8a1f9738-d7b7-42b5-bf61-c9cd7e9a1f76 [0xF8A1] (Lonsonho  Switch 1)>
2022-06-17T10:52:41.916823451+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 8a1f9738-d7b7-42b5-bf61-c9cd7e9a1f76 [0xF8A1] (Lonsonho  Switch 1)>
2022-06-17T10:52:41.925511118+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 8a1f9738-d7b7-42b5-bf61-c9cd7e9a1f76 [0xF8A1] (Lonsonho  Switch 1)>
2022-06-17T10:52:41.946223785+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 8a1f9738-d7b7-42b5-bf61-c9cd7e9a1f76 [0xF8A1] (Lonsonho  Switch 1)>
2022-06-17T10:52:41.967406118+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 8a1f9738-d7b7-42b5-bf61-c9cd7e9a1f76 [0xF8A1] (Lonsonho  Switch 1)>
2022-06-17T10:52:41.990855118+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 8a1f9738-d7b7-42b5-bf61-c9cd7e9a1f76 [0xF8A1] (Lonsonho  Switch 1)>
2022-06-17T10:52:42.007830785+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 8a1f9738-d7b7-42b5-bf61-c9cd7e9a1f76 [0xF8A1] (Lonsonho  Switch 1)>
2022-06-17T10:52:42.021802785+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 8a1f9738-d7b7-42b5-bf61-c9cd7e9a1f76 [0xF8A1] (Lonsonho  Switch 1)>
2022-06-17T10:52:42.036738118+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 8a1f9738-d7b7-42b5-bf61-c9cd7e9a1f76 [0xF8A1] (Lonsonho  Switch 1)>
2022-06-17T10:52:42.059670118+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch : false
2022-06-17T10:52:42.066920785+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 8a1f9738-d7b7-42b5-bf61-c9cd7e9a1f76 [0xF8A1] (Lonsonho  Switch 1)>
2022-06-17T10:52:42.074243785+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 8a1f9738-d7b7-42b5-bf61-c9cd7e9a1f76 [0xF8A1] (Lonsonho  Switch 1)>
2022-06-17T10:52:42.087139451+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 8a1f9738-d7b7-42b5-bf61-c9cd7e9a1f76 [0xF8A1] (Lonsonho  Switch 1)>
2022-06-17T10:52:42.093965118+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 8a1f9738-d7b7-42b5-bf61-c9cd7e9a1f76 [0xF8A1] (Lonsonho  Switch 1)>
2022-06-17T10:52:42.107036785+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 8a1f9738-d7b7-42b5-bf61-c9cd7e9a1f76 [0xF8A1] (Lonsonho  Switch 1)>
2022-06-17T10:52:42.115022118+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 8a1f9738-d7b7-42b5-bf61-c9cd7e9a1f76 [0xF8A1] (Lonsonho  Switch 1)>
2022-06-17T10:52:42.127002451+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 8a1f9738-d7b7-42b5-bf61-c9cd7e9a1f76 [0xF8A1] (Lonsonho  Switch 1)>
2022-06-17T10:52:42.132922118+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 8a1f9738-d7b7-42b5-bf61-c9cd7e9a1f76 [0xF8A1] (Lonsonho  Switch 1)>
2022-06-17T10:52:42.146899785+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 8a1f9738-d7b7-42b5-bf61-c9cd7e9a1f76 [0xF8A1] (Lonsonho  Switch 1)>
2022-06-17T10:52:42.159422118+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 8a1f9738-d7b7-42b5-bf61-c9cd7e9a1f76 [0xF8A1] (Lonsonho  Switch 1)>
2022-06-17T10:52:42.166616785+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: 8a1f9738-d7b7-42b5-bf61-c9cd7e9a1f76 [0xF8A1] (Lonsonho  Switch 1)>
[... cut ...]
2022-06-17T10:57:09.935172143+00:00 INFO zigbee-tuya-switch  <ZigbeeDevice: 8a1f9738-d7b7-42b5-bf61-c9cd7e9a1f76 [0xF8A1] (Lonsonho  Switch 1)> emitting event: {"attribute_id":"switch","capability_id":"switch","component_id":"main","state":{"value":"off"}}
2022-06-17T10:57:09.951819143+00:00 DEBUG zigbee-tuya-switch  Lonsonho  Switch 1 device thread event handled
2022-06-17T10:57:14.442741145+00:00 DEBUG zigbee-tuya-switch  driver device thread event handled
2022-06-17T10:57:44.457451160+00:00 DEBUG zigbee-tuya-switch  driver device thread event handled
2022-06-17T10:58:14.467015841+00:00 DEBUG zigbee-tuya-switch  driver device thread event handled
listening for logs...

obmaz, 2022-06-20:
로그 추가하여 배포하였습니다.
다시 테스트 부탁 드립니다.

init 시점 로그가 없어서, 관련 부분 로그가 필요합니다. (<<---- Moon ---->> single / device_init)
cli에서 로그 실행 후 설치 테스트 부탁 드립니다.

obmaz, 2022-06-21:
지금 해주시는것과 동일 합니다. 로그찍히는 순간 이후부터 페어링 부탁 드립니다. 지금 공유해주신건 페어링 하는 부분 로그가 안 나와 있어서요.

obmaz, 2022-06-21:
로그 감사합니다.
의심 부분 수정 후 배포하였습니다.

7destiny, 2022-06-22:
계속 신경써주셔서 감사합니다. 드라이버 삭제후 재페어링 해보아도 아직 2분후 꺼집니다 ㅠ

7destiny, 2022-06-24:
참고로 2구 스위치에선 2분후 꺼지는 현상이 없네요

obmaz, 2022-06-25:
1구 2구 페어링 부터 하는 로그 확보 가능 하면 부탁 드립니다.

7destiny, 2022-06-25:
2구 페어링
│ 10 │ de2773fd-2901-4254-9009-276f1ca90350 │ zigbee-tuya-switch             │
└────┴──────────────────────────────────────┴────────────────────────────────┘
? Select a driver. 10
2022-06-25T10:30:24.713792247+00:00 TRACE zigbee-tuya-switch  Setup driver zigbee-tuya-switch with lifecycle handlers:
DeviceLifecycleDispatcher: zigbee-tuya-switch
  default_handlers:
    driverSwitched:
    doConfigure:
    init:
  child_dispatchers:
    DeviceLifecycleDispatcher: zigbee-tuya-switch -> mutil switch
      default_handlers:
        infoChanged:
        driverSwitched:
        doConfigure:
        added:
        init:
      child_dispatchers:

2022-06-25T10:30:24.721728247+00:00 TRACE zigbee-tuya-switch  Setup driver zigbee-tuya-switch with Capability handlers:
CapabilityCommandDispatcher: zigbee-tuya-switch
  default_handlers:
    refresh:
      refresh
    switch:
      off
      on
  child_dispatchers:
    CapabilityCommandDispatcher: zigbee-tuya-switch -> mutil switch
      default_handlers:
        switch:
          off
          on
      child_dispatchers:

2022-06-25T10:30:24.728580913+00:00 TRACE zigbee-tuya-switch  Setup driver zigbee-tuya-switch with Zigbee handlers:
ZigbeeMessageDispatcher: zigbee-tuya-switch
  default_handlers:
    attr:
      ZclClusterAttributeValueHandler: cluster: OnOff, attribute: OnOff
    global:
      ZclGlobalCommandHandler: cluster: OnOff, command: DefaultResponse
    cluster:
    zdo:
  child_dispatchers:
    ZigbeeMessageDispatcher: zigbee-tuya-switch -> mutil switch
      default_handlers:
        attr:
          ZclClusterAttributeValueHandler: cluster: OnOff, attribute: OnOff
        global:
        cluster:
        zdo:
      child_dispatchers:

2022-06-25T10:30:24.788687580+00:00 TRACE zigbee-tuya-switch  Zigbee Device: f1dbd058-63d6-4d63-9877-41f4affd4836
Manufacturer: _TZ3000_fvh3pjaz Model: TS0012
        [1]: Basic, Groups, Scenes, OnOff       [2]: Groups, Scenes, OnOff
2022-06-25T10:30:24.797421247+00:00 TRACE zigbee-tuya-switch  Received event with handler _resync
2022-06-25T10:30:24.802607913+00:00 TRACE zigbee-tuya-switch  Received event with handler environment_info
2022-06-25T10:30:24.814985580+00:00 DEBUG zigbee-tuya-switch  driver device thread event handled
2022-06-25T10:30:24.825883580+00:00 TRACE zigbee-tuya-switch  Received event with handler environment_info
2022-06-25T10:30:24.831197580+00:00 DEBUG zigbee-tuya-switch  Z-Wave hub node ID environment changed.
2022-06-25T10:30:24.837706247+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: f1dbd058-63d6-4d63-9877-41f4affd4836 [0x49D6] (Lonsonho Switch 2)>
2022-06-25T10:30:24.843449913+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: f1dbd058-63d6-4d63-9877-41f4affd4836 [0x49D6] (Lonsonho Switch 2)>
2022-06-25T10:30:24.879631580+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: f1dbd058-63d6-4d63-9877-41f4affd4836 [0x49D6] (Lonsonho Switch 2)>
2022-06-25T10:30:24.884889580+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch : true / device.fingerprinted_endpoint_id :     1
2022-06-25T10:30:24.890559247+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: f1dbd058-63d6-4d63-9877-41f4affd4836 [0x49D6] (Lonsonho Switch 2)>
2022-06-25T10:30:24.895852580+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: f1dbd058-63d6-4d63-9877-41f4affd4836 [0x49D6] (Lonsonho Switch 2)>
2022-06-25T10:30:24.903831247+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: f1dbd058-63d6-4d63-9877-41f4affd4836 [0x49D6] (Lonsonho Switch 2)>
[... 1021 lines cut ...]
2022-06-25T10:43:25.764196280+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: aca45099-670d-4c6a-bad7-05c6ea25c02a [0x618D] (Lonsonho  Switch 1)>
2022-06-25T10:43:25.769521613+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: aca45099-670d-4c6a-bad7-05c6ea25c02a [0x618D] (Lonsonho  Switch 1)>
2022-06-25T10:43:25.774982280+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: aca45099-670d-4c6a-bad7-05c6ea25c02a [0x618D] (Lonsonho  Switch 1)>
2022-06-25T10:43:25.781011613+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: aca45099-670d-4c6a-bad7-05c6ea25c02a [0x618D] (Lonsonho  Switch 1)>
2022-06-25T10:43:25.786575280+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch : false
2022-06-25T10:43:25.797892280+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: aca45099-670d-4c6a-bad7-05c6ea25c02a [0x618D] (Lonsonho  Switch 1)>
2022-06-25T10:43:25.805249280+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: aca45099-670d-4c6a-bad7-05c6ea25c02a [0x618D] (Lonsonho  Switch 1)>
2022-06-25T10:43:25.816775947+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: aca45099-670d-4c6a-bad7-05c6ea25c02a [0x618D] (Lonsonho  Switch 1)>
2022-06-25T10:43:25.827954613+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch :       <ZigbeeDevice: aca45099-670d-4c6a-bad7-05c6ea25c02a [0x618D] (Lonsonho  Switch 1)>
[... cut ...]
2022-06-25T10:43:26.172324947+00:00 INFO zigbee-tuya-switch  <ZigbeeDevice: aca45099-670d-4c6a-bad7-05c6ea25c02a [0x618D] (Lonsonho  Switch 1)> received Zigbee message: < ZigbeeMessageRx || type: 0x00, < AddressHeader || src_addr: 0x618D, src_endpoint: 0x01, dest_addr: 0x0000, dest_endpoint: 0x01, profile: 0x0104, cluster: OnOff >, lqi: 0xE8, rssi: -42, body_length: 0x0007, < ZCLMessageBody || < ZCLHeader || frame_ctrl: 0x18, seqno: 0x04, ZCLCommandId: 0x0A >, < ReportAttribute || < AttributeRecord || AttributeId: 0x0000, DataType: Boolean,

7destiny, 2022-06-25:
몇일전엔 안되었는데 오늘은 다르게 기존에 2,3구로 사용하던 다른 엣지를 다지우고 다시 설치후 2구부터 페어링하고 1구 페어링을 했는데 1구가 이제 2분뒤 안꺼집니다! 근대 페어링시 1,2구에 똑같은 에러가 보이네요. error: [string "st/dispatcher.lua"]:229 
참고하시고 일단 2분뒤 꺼지지는 않습니다~ 감사드리며 혹여나 위 로그 참고하시길 바랍니다.

7destiny, 2022-06-28:
스위치 전원이 가끔씩 순간적으로 나갔다 들어오는 문제로 스위치를 예비로 사놓았던게 있는데 이 스위치도 2분마다 꺼지네요 ㅠ 3버튼이거 이미 핑거프린트는 등록이 된건데 왜 이전껀 문제가 없었는게 이건 또 왜그런건지;; _TZ3000_wyhuocal 도 2분 오프안되게 적용 부탁드립니다 ㅠ

7destiny, 2022-07-11:
스위치가 많은데다 매번 edge랑 dth 바꾸면서 자동화 다시 손보고해야되서 집사람 등짝 스메싱으로 테스트가 늦었습니다…일단 기존에 문제없이 사용중인 dth는 놔두고 edge 설치후 스위치 삭제후 페어링하여 edge로 잡히기에 테스트시 안꺼지는게 아니라 2분뒤 순간꺼졌다가 켜집니다…1갱,2갱까지만 테스트 했네요 ㅠ

7destiny, 2022-07-11:
2갱 스위치 페어링후 메인 스위치 자동으로 켜지기에 스위치끄고 다시 켜서 2분뒤 순간 꺼졌다 켜질때까지의 로그

? Select a driver. 9
2022-07-11T12:41:48.465893981+00:00 TRACE zigbee-tuya-switch  Setup driver zigbee-tuya-switch with lifecycle handlers:
DeviceLifecycleDispatcher: zigbee-tuya-switch
  default_handlers:
    init:
    driverSwitched:
    doConfigure:
  child_dispatchers:
    DeviceLifecycleDispatcher: zigbee-tuya-switch -> mutil switch
      default_handlers:
        infoChanged:
        added:
        doConfigure:
        init:
        driverSwitched:
      child_dispatchers:

2022-07-11T12:41:48.472441981+00:00 TRACE zigbee-tuya-switch  Setup driver zigbee-tuya-switch with Capability handlers:
CapabilityCommandDispatcher: zigbee-tuya-switch
  default_handlers:
    switch:
      on
      off
    refresh:
      refresh
  child_dispatchers:
    CapabilityCommandDispatcher: zigbee-tuya-switch -> mutil switch
      default_handlers:
        switch:
          on
          off
      child_dispatchers:

2022-07-11T12:41:48.479955314+00:00 TRACE zigbee-tuya-switch  Setup driver zigbee-tuya-switch with Zigbee handlers:
ZigbeeMessageDispatcher: zigbee-tuya-switch
  default_handlers:
    attr:
      ZclClusterAttributeValueHandler: cluster: OnOff, attribute: OnOff
    global:
      ZclGlobalCommandHandler: cluster: OnOff, command: DefaultResponse
    cluster:
    zdo:
  child_dispatchers:
    ZigbeeMessageDispatcher: zigbee-tuya-switch -> mutil switch
      default_handlers:
        attr:
          ZclClusterAttributeValueHandler: cluster: OnOff, attribute: OnOff
        global:
        cluster:
        zdo:
      child_dispatchers:

2022-07-11T12:41:48.529886314+00:00 TRACE zigbee-tuya-switch  Zigbee Device: 39c99f1b-ca00-4a91-aab0-f31b5d230b4b
Manufacturer: _TZ3000_fvh3pjaz Model: TS0012
        [1]: Basic, Groups, Scenes, OnOff       [2]: Groups, Scenes, OnOff
2022-07-11T12:41:48.537445647+00:00 DEBUG zigbee-tuya-switch  driver device thread event handled
2022-07-11T12:41:48.543132647+00:00 TRACE zigbee-tuya-switch  Received event with handler _resync
2022-07-11T12:41:48.550988314+00:00 TRACE zigbee-tuya-switch  Received event with handler environment_info
2022-07-11T12:41:48.573354647+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch : true / device.fingerprinted_endpoint_id :     1
2022-07-11T12:41:48.590213314+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch : true / device.fingerprinted_endpoint_id :     1
2022-07-11T12:41:48.603056647+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / is_multi_switch : true / device.fingerprinted_endpoint_id :     1
2022-07-11T12:41:48.610068981+00:00 TRACE zigbee-tuya-switch  Found DeviceLifecycleDispatcher handler in zigbee-tuya-switch -> mutil switch
2022-07-11T12:41:48.621181981+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / device_init
2022-07-11T12:41:48.627183981+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> is_120sec_issue : test device:get_manufacturer() :      _TZ3000_fvh3pjaz
2022-07-11T12:41:48.634943981+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> is_120sec_issue : test device:get_model() :     TS0012
2022-07-11T12:41:48.640587647+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> is_120sec_issue : test IS_120SEC_ISSUE_FINGERPRINTS.mfr :
2022-07-11T12:41:48.652731647+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> is_120sec_issue : test IS_120SEC_ISSUE_FINGERPRINTS.mfr :
2022-07-11T12:41:48.691469981+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> is_120sec_issue : true / device.fingerprinted_endpoint_id :     1
[... 174 lines cut ...]
2022-07-11T12:44:56.568178219+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / attr_handler :  switch1
2022-07-11T12:44:56.574539219+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / remapSwitch
2022-07-11T12:44:56.590126219+00:00 INFO zigbee-tuya-switch  <ZigbeeDevice: 39c99f1b-ca00-4a91-aab0-f31b5d230b4b [0xDECF] (Lonsonho Switch 2)> emitting event: {"attribute_id":"switch","capability_id":"switch","component_id":"main","state":{"value":"on"}}
2022-07-11T12:44:56.610437886+00:00 INFO zigbee-tuya-switch  <ZigbeeDevice: 39c99f1b-ca00-4a91-aab0-f31b5d230b4b [0xDECF] (Lonsonho Switch 2)> emitting event: {"attribute_id":"switch","capability_id":"switch","component_id":"switch1","state":{"value":"on"}}
2022-07-11T12:44:56.644010553+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / remapSwitch
2022-07-11T12:44:56.652228886+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / syncMainComponent :     switch1
2022-07-11T12:44:56.688339886+00:00 INFO zigbee-tuya-switch  <ZigbeeDevice: 39c99f1b-ca00-4a91-aab0-f31b5d230b4b [0xDECF] (Lonsonho Switch 2)> emitting event: {"attribute_id":"switch","capability_id":"switch","component_id":"main","state":{"value":"on"}}
2022-07-11T12:44:56.706067553+00:00 DEBUG zigbee-tuya-switch  Lonsonho Switch 2 device thread event handled
2022-07-11T12:44:56.719959553+00:00 TRACE zigbee-tuya-switch  Received event with handler zigbee
2022-07-11T12:44:56.732523886+00:00 INFO zigbee-tuya-switch  <ZigbeeDevice: 39c99f1b-ca00-4a91-aab0-f31b5d230b4b [0xDECF] (Lonsonho Switch 2)> received Zigbee message: < ZigbeeMessageRx || type: 0x00, < AddressHeader || src_addr: 0xDECF, src_endpoint: 0x02, dest_addr: 0x0000, dest_endpoint: 0x01, profile: 0x0104, cluster: OnOff >, lqi: 0xFF, rssi: -32, body_length: 0x0011, < ZCLMessageBody || < ZCLHeader || frame_ctrl: 0x08, seqno: 0x01, ZCLCommandId: 0x0A >, < ReportAttribute || < AttributeRecord || At [... cut ...]
2022-07-11T12:44:56.843242886+00:00 INFO zigbee-tuya-switch  <<---- Moon ---->> multi / syncMainComponent :     switch1
2022-07-11T12:44:56.850263553+00:00 INFO zigbee-tuya-switch  <ZigbeeDevice: 39c99f1b-ca00-4a91-aab0-f31b5d230b4b [0xDECF] (Lonsonho Switch 2)> emitting event: {"attribute_id":"switch","capability_id":"switch","component_id":"main","state":{"value":"on"}}
2022-07-11T12:44:56.889605886+00:00 DEBUG zigbee-tuya-switch  Lonsonho Switch 2 device thread event handled
2022-07-11T12:45:09.546922892+00:00 TRACE zigbee-tuya-switch  Received event with handler zigbee
listening for logs... /

obmaz, 2022-07-12:
답변 감사합니다. 로그에 오류가 보여서 수정 하였습니다.
시간 되시면 테스트 부탁 드립니다. (2022-07-12T10:12:58.509335315)

7destiny, 2022-08-18:
테스트가 늦었습니다; 역시나 2분뒤 잠깐 껌뻑~ 꺼진뒤 바로 켜지네요 ㅠ

obmaz, 2023-11-23:
더 이상은 테스트가 힘들어 일단 issue 종료 하겠습니다.
추후 관련 기기를 테스트 할 수 있으면 reopen해서 추가 코멘트 남기겠습니다.

closed 2023-11-23 by obmaz