obmaz/smartthings_edge_driver #21: Add Tuya 1 Button - Please

dotanshai, 2023-03-23:
Hi, 
Can you please add the following 1 Button:

fingerprint
```
inClusters: "0000,0001,0003,0004,0006,1000,E001",
outClusters: "0019,000A,0003,0004,0006,0008,1000",
manufacturer: "_TZ3000_ja5osu5g", 
model: "TS004F", 
deviceJoinName: "MOES Smart Button (ZT-SY-SR-MS)" 

```
Thanks a lot... 👍

obmaz, 2023-03-24:
The FP had been added and released (2023-03-24T04:25:32.802655099)
Let me know whether it work

dotanshai, 2023-03-29:
Please ignore pervious log file. This file contain pairing and some presses of the button.
I hope this can help 
https://drive.google.com/file/d/1XCVUzgYSkk8obiucyocGxmki0cxW8hKB/view?usp=share_link

dotanshai, 2023-04-18:
Your driver WORKS !! 
But need some adjustments. Please that this button have two modes : ON/OFF and Events (Presses)
You can switch between modes by configure Attribute 0x8004 in Cluster 0x06 or by x3 presses as it explain in the user manual
[image]
I used the presses since by configuring the device is not yet implemented by you....
One more thing , while in ON/OFF mode. x2 presses equal on, x1 press equal OFF

Here are some more logs, showing that presses , On/Off
https://drive.google.com/file/d/1Cug21BrdZHVredRANgQuDSqp2YOvGVfB/view?usp=sharing

Here is the manual of that device if needed
https://drive.google.com/file/d/1YEpa_T8W-0WpdYjX86yZLefgEZx-XvvZ/view?usp=sharing

obmaz, 2023-11-23:
Thanks for sharing about modes.
It's hard to implement right now, but I'll put it to the backlog and try to implement in the feature.

closed 2023-11-23 by obmaz