obmaz/smartthings_edge_driver #23: Can you please add

dotanshai, 2023-11-20:
Can you please add:
**Manufacturer:_TZ3000_mrpevh8p
Model: TS0041**
FingerPrinted_EndPoint.Id: 0x01
App Version: 0x43
ZCL Version: 0x03
Network ID: 0x9C70
Zigbee EUI: A4C1388201373D32
Device ID: e758aba8-471a-48ad-b326- a886e30dd901
Server Cluster:
Ep: 0x03={0006,0001,0006}
Ep: 0x02={0006,0001,0006 }
Ep: 0x01={0001,0006, E000,0000 }
Ep: 0x04={0006,0001,0006}
Client Clusters:
Ep: 0x01={0019,000A }

This is very similar to (or even identical) to a driver you already develop:
  - id: _TZ3000_tk3s5tyg/TS0041
Please help

obmaz, 2023-11-23:
The FP had been added and deployed (2023-11-23T03:11:46.107661515)

closed 2023-11-23 by obmaz

dotanshai, 2023-11-24:
Works great can be close. THANKS!!