obmaz/smartthings_edge_driver #24: Can you please add 6 button scene switch - TZ3000_iszegwpd

dotanshai, 2023-11-24:
Zigbee Switch

Device Info
**Manufacturer: _TZ3000_iszegwpd
Model: TS0046**
FingerPrinted_EndPoint.Id: 0x01
App Version: Ox43
ZCL Version: 0x03
Network ID: 0xD612
Zigbee EUI: A4C13889D300131A
Device ID: 6b09c704-ec92-41a2-af75-
3d97fa5a2d34
:Server Cluster
Ep: 0x03={0006,0001,0006}
Ep: 0x04-0006,0001,0006}
Ep: 0x01={0001,0006,000,0000}
Ep: 0x02={0006,0001,0006}
:Client Clusters
Ep: 0x01={0019,000A }

obmaz, 2023-11-29:
Unfortunately, 6 buttons are not supported.

closed 2023-11-29 by obmaz

dotanshai, 2024-04-05:
this was closed 6 months ago... Anything new ? Do you have plans to support this 6 buttons scene switch ?

obmaz, 2024-04-05:
Additionally, "_TZ3000_iszegwpd" was added about two months ago at the request of the Korean community, and was confirmed to work.