obmaz/smartthings_edge_driver #25: Multi-switch 4 gang

jonicunha, 2023-12-14:
Hi,

Can you please create the driver for this device?

It’s a remote control with 4 switchs, each one with single, double and long press function.

Manufacturer: _TZ3000_wbfgbpxq
Model: TS0044
FingerPrinted_EndPoint.Id: 0x01
App Version: 0x43
ZCL Version: 0x03
Network ID: 0xFEB6
Zigbee EUI: A4C1385F7F675011
Device ID: e868e50c-d425-4d43-b0a0-10f64445220e

Server Cluster:
Ep: 0x02={ 0006,0001,0006 }
Ep: 0x01={ 0001,0006,E000,0000 }
Ep: 0x04={ 0006,0001,0006 }
Ep: 0x03={ 0006,0001,0006 }

Client Clusters:
Ep: 0x01={ 0019,000A }

Signal Metrics:
LQI: 144 … RSSI: -62 dbm

Thank you!

obmaz, 2023-12-15:
@jonicunha 
Added FP and applied. (2023-12-15T02:45:47.720164129)
Let me know whehter it works.

jonicunha, 2023-12-15:
Hi @obmaz 

Thanks for your quick reply!
It is working for pressing and double pressing, but not for holding.
It's not an important function and it's ok as it is but if you could take a look I'd appreciate it!

I have a question: Should the status always be connected? I have other buttons and the status is on standby.

Once again thank you so much! How can I gift you? Do you have PayPal or something?

obmaz, 2023-12-19:
> I have a question: Should the status always be connected? I have other buttons and the status is on standby.

Since the battery, not the button, is shown in the main state, it is probably marked as connected.

> Once again thank you so much! How can I gift you? Do you have PayPal or something?

You shouldn't have, I'll have only your mind.

closed 2023-12-19 by obmaz