obmaz/smartthings_edge_driver #27: Can you please add the following button _TZ3000_yj6k7vfo

dotanshai, 2024-01-16:
Manufacturer: TZ3000_yj6k7vfo
Model: TS0041

[image]

Thanks,

obmaz, 2024-01-17:
Done (2024-01-17T01:09:52.510373885)

closed 2024-01-17 by obmaz

dotanshai, 2024-01-18:
There is a problem with manufacturer p/n. In your driver : TZ3000_yj6k7vfo , while the actual should be : **_**TZ3000_yj6k7vfo (starting underscore is missing), can you please check ? 
Thanks....

obmaz, 2024-01-19:
> There is a problem with manufacturer p/n. In your driver : TZ3000_yj6k7vfo , while the actual should be : _TZ3000_yj6k7vfo (starting underscore is missing), can you please check ? Thanks....

It's my mistake.
Fixed (2024-01-19T02:08:24.845871097)

dotanshai, 2024-01-21:
After your fixing, All works. Thank you for your work  👍 😄