obmaz/smartthings_edge_driver #28: Can you please add Ikea RODRET Dimmer ?

dotanshai, 2024-02-10:
Can you please add Ikea RODRET Dimmer here are some more details:
[image]

Another user already added this device, but didn't publish his work in a private channel. Read also here:
https://community.smartthings.com/t/post-requests-for-zigbee-edge-drivers-here-community-created/262081/1454

obmaz, 2024-02-17:
Does this mean you want to add that finger print to the Tuya driver?

dotanshai, 2024-02-17:
Tuya or Ikea , doesn't really matters for me. As long as the button will work.
Is that possible?

obmaz, 2024-02-20:
I found the channel that support Ikea RODRET Dimmer. Use below channel
-  https://bestow-regional.api.smartthings.com/invite/r3Myde4gbWjp (Zigbee Button)

closed 2024-02-20 by obmaz