obmaz/smartthings_edge_driver #30: _TZ3000_an5rjiwd

ryanpun1223, 2024-03-13:
[image]
Can we add _TZ3000_an5rjiwd to the driver ?

thanks !

obmaz, 2024-03-13:
Is it a switch or button? This is because TS0041 is generally a button, but if you look at the screenshot, it is a generic switch. If it is a switch, 4-button switch is not supported.

ryanpun1223, 2024-03-13:
It is a button. It's just that for some reason it was recognized as a
switch.

This is the button.

https://www.aliexpress.com/item/1005006124399376.html?spm=a2g0o.order_list.order_list_main.41.67901802LTJN2u

Regards,

Ryan Pun

+31 (0)6 5891 4293
+1 (734) 846 - 8218

closed 2024-04-05 by obmaz