obmaz/smartthings_edge_driver #32: can't assign your driver to my tuya zigbee button

oualibadr, 2024-04-24:
Hi Obmaz,

I hope you are doing well,

I was looking for an edge driver for my tuya zigbee button when I found a post (https://community.smartthings.com/t/is-there-a-zigbee-smart-button-driver-driver/277743) on the smatthings forum mentionning that someone solved the issue by installing your drivers. so I did the same.

My problem is that I can't assign your driver to the device: I can't see it in the dropdown drivers list 

[image]
[image]

Any idea?

Thanks,

oualibadr, 2024-04-24:
there you go : _TZ3000_qgwcxxws

obmaz, 2024-04-24:
I also need model number like 'TS00nn'

oualibadr, 2024-04-24:
here it is TS0041

obmaz, 2024-04-25:
The FP was added and applied (2024-04-25T01:17:51.351320819)

closed 2024-04-25 by obmaz

oualibadr, 2024-04-25:
Many Thanks !

oualibadr, 2024-04-26:
Working like a charm :)