obmaz/smartthings_edge_driver #36: 핑거프린트 추가요청합니다.

GSL-R, 2024-06-28:
알리에서 구매한 4버튼인데, 핑거프린트가 등록되어 있지 않아서인지 드라이버가 보이질 않네요. 구글링해도 결과가 없어 혹시 가능하실까 싶어 추가요청합니다. 감사합니다.

https://a.aliexpress.com/_oCp8tWw 

Manufacturer Code: _TZ3000_0ht8dnxj
Model: TS004F
[image]
[image]

obmaz, 2024-06-28:
추가 하여 배포 하였습니다. (2024-06-28T11:08:49.708682556)

closed 2024-06-28 by obmaz

GSL-R, 2024-06-28:
빠른 대응 감사합니다! 드라이버 지웠다 재설치하니 바로 뜨네요
그런데 버튼에 루틴 할당 후 눌러도 반응이 없습니다. 기록에도 아무것도 뜨질 않네요. 기기 지우고 다시 연결도 해봤는데도 동일합니다.
[image]
[image]

GSL-R, 2024-06-28:
답변 감사합니다! 찾아보겠습니다

GSL-R, 2024-06-28:
설명서를 찾아보니 디밍모드와 씬 모드 전환에 대한 설명이 있어 바꿔보니까 잘 작동합니다! 3-4번 버튼을 10초이상 누르고 있으니까 점멸하면서 모드가 변경되네요. 이후엔 버튼 모두 잘 반응합니다.

막상 구매하고 나니 핑거프린트가 없는 모델이라 막막했는데 덕분에 잘 사용할 수 있을거 같아요! 감사합니다.