obmaz/smartthings_edge_driver #37: Help the add a fingerprint of the schnider electric wiser freelocate 4k switch

murali-dotworld, 2024-07-04:
Hello i have a wiser freelocate 4k switch which is similar to the Moes 4 gang switch.. i have a fingerprint of the device.. can you help me add a fingerprint into your driver. i am new to the smartthings, so i dont know how to add the fingerprint.. can you help us..
[image]

obmaz, 2024-07-04:
I added FP and applied (2024-07-04T07:25:51.325330947)
I don't know if it will work because the format is different from Tuya. Please let me know if it works.

murali-dotworld, 2024-07-04:
Hello i tried adding the device to driver,but it not showing your driver inside my devices...

murali-dotworld, 2024-07-04:
Can you change a device label and manufacture...?? swap both.. that the problem i think so

obmaz, 2024-07-04:
I changed manufacture and model (2024-07-04T07:50:16.808653239)
Device Label is not a fingerprint that is for Smartthings UI

murali-dotworld, 2024-07-04:
Thanks for your help... the device in your driver added successfully... Only the battery is updating... not getting a switch status...

obmaz, 2024-07-04:
If the fingerprint is correct, the driver matches. However, if the function does not work, it is because the zigbee command rules are different.

murali-dotworld, 2024-07-04:
okk thank you....

closed 2024-07-04 by obmaz