obmaz/smartthings_edge_driver #48: Can you please add 4 Keys scene switch _TZ3000_et7afzxz

gilrose1, 2024-09-01:
Hi ,

Can you please the following scene swithc to your Edge Driver :

```
Manufacturer: _TZ3000_et7afzxz
Model: TS004F
FingerPrinted_EndPoint.Id: 0x01
App Version: Unknown
ZCL Version: Unknown
Network ID: 0x8DB2
Zigbee EUI: 881A14FFFEB6437B
Device ID: 8d7b8063-82bf-4d51-923a-16486e3a3da1

Server Cluster:
Ep: 0x02={ 0001,0003,0004,0006,1000 }
Ep: 0x03={ 0001,0003,0004,0006,1000 }
Ep: 0x04={ 0001,0003,0004,0006,1000 }
Ep: 0x01={ 0000,0001,0003,0004,0006,1000 }

Client Clusters:
Ep: 0x02={ 0003,0004,0005,0006,0008,1000 }
Ep: 0x03={ 0003,0004,0005,0006,0008,1000 }
Ep: 0x04={ 0003,0004,0005,0006,0008,1000 }
Ep: 0x01={ 0019,000A,0003,0004,0005,0006,0008,
1000 }
```

Thanks,
Gil

obmaz, 2024-09-02:
Applied (2024-09-02T02:28:32.499296633)

closed 2024-09-02 by obmaz

gilrose1, 2024-09-02:
Driver is not publish . I'm still getting the previus version. See in CLI

[image]

Can you please re-publish ?

Gil

obmaz, 2024-09-03:
I published it again (2024-09-03T01:08:52.117570193)
It is usually reflected in the hub within 48 hours. If you want it to be updated immediately, you can uninstall and then reinstall the driver, and the latest version will be installed on the hub.

dotanshai, 2024-09-03:
We have some issues with that device. According to CLI nothing is sent to LOG while pressing. In addition the device control other devices without adding any routine. Seems like it have a binding mode, like Ikea devices. Is it possible to configure during the pairing to force normal Zigbee mode ?