obmaz/smartthings_edge_driver #49: Zemismart 4 button tuya zigbee

Omer-harel, 2024-10-03:
Hello,

Thank you for your help.
Month ago we requested to develop driver for tuya zemismart 4 button (zigbee).
After a while it stop working and saw battary still 100 and show connected.

Since then I replace 3 batteries.
Aftel a while its stop working. 
Maybe it drains battary some how.

Is it pobbile For you to look at the driver. Maybe change settings  to prevent from happening?

Thank you.

Omer-harel, 2024-10-03:
here is input from log
[obmaz.txt](https://github.com/user-attachments/files/17242121/obmaz.txt)

obmaz, 2024-10-03:
As of now, it is not possible. In the case of Tuya, some devices have battery drain, while others do not, even if they are the same device.

closed 2024-10-03 by obmaz