obmaz/smartthings_edge_driver #50: Legrand Valena with Netatmo

TriXso, 2024-11-14:
Is it possible to add this device to your driver?

Here is cluster documentation from Legrand:
[NLTOO-cluster-list.xlsx](https://github.com/user-attachments/files/17747752/NLTOO-cluster-list.xlsx)

Manufacturer: Legrand
Model: Remote switch
FingerPrinted_EndPoint.Id: 0x01
App Version: Unknown
ZCL Version: Unknown
Network ID: 0x12D8
Zigbee EUI: 0004740000C44E37
Device ID: 5517dc8a-93a9-4773-93ee-f26d3fdf64ef

Server Cluster:
Ep: 0x01={ 0000,0003,000F,0020,0001,FC01 }

Client Clusters:
Ep: 0x01={ 0003,0006,0008,0000,FC01,0019 }

Signal Metrics:

obmaz, 2024-11-14:
Need more information.
1. Button or Switch
2. How many button (switch)

TriXso, 2024-11-14:
This is the board. It has 2 buttons. One labeled on, other one off. Third one is for reset/pairing.
[image]

obmaz, 2024-11-14:
version (2024-11-14T12:32:15.636462592)

I added the fingerprint and deployed it. If you disconnect and reconnect the driver channel, the latest version should be applied. Since I only added the code without testing on an actual device, I’m not sure if it will work as expected. Please let me know if it functions properly.

dotanshai, 2024-11-15:
In order to use obmaz driver , you need to install the driver from his channel. His channel is here:
https://api.smartthings.com/invitation-web/accept?id=2d6da83b-ae22-4fda-8af5-bbb8890872d4

install zigbee tuya button driver in your case.

TriXso, 2024-11-15:
[image]
That's what i did. Hub is showing that driver is installed but it's not possible to choose it for the device.

obmaz, 2024-11-23:
I changed model name and redeployed. (2024-11-23T14:44:22.782059745)

I would appreciate it if you could test it and let me know if it works.

obmaz, 2024-11-26:
It seems like it doesn't work because the signal specifications are different. It seems like it will be difficult to support it further without the actual device.

closed 2024-11-26 by obmaz