obmaz/smartthings_edge_driver #52: Support for push/rotary dimmer switch from Tuya

Staffed24, 2024-11-23:
Hi,

I have this button / dimmer from tuya, which can be rotated and pushed: https://www.aliexpress.com/item/1005007720682666.html

Manufacturer code _TZ3000_402vrq2i
Model TS004F

Have tried a few drivers but haven’t found any driver that shows up in the driver list for the device. Could you add support for this device?
[image]

obmaz, 2024-11-23:
I added FP and deployed (2024-11-23T14:33:27.533229293)

The TS004F model has a scene mode, which might cause it to not function properly. If it does not work as expected, please try switching the mode and testing it again. Since the method to change modes varies by device, it may be difficult to provide specific instructions.

Staffed24, 2024-11-24:
Thanks! Will try and report back! :)

Staffed24, 2024-11-24:
The push button works fine now. Thanks a lot!!! 

The rotation does not seem to work however. I guess mode switching is done on the device through some sort of combination of button pushes/long press?

If I push the button three times in a quick sequence, the indicator light on the bottom lights up for about three seconds, but the rotary switch still won’t work. I’ve tried pushing the reset switch up to 10 times in a sequence and holding it for three seconds (so it doesn’t reset) and holding/pushing the button and rotating the rotary switch.

Could it be that the rotary switch doesn’t work in either mode? I’m happy that the button works, so it doesn’t matter that much if we can’t get the rotary switch working :)

obmaz, 2024-11-26:
It seems like it doesn't work because the signal specifications are different. It seems like it will be difficult to support it further without the actual device.

closed 2024-11-26 by obmaz

marcos-mvs, 2025-02-08:
> It seems like it doesn't work because the signal specifications are different. It seems like it will be difficult to support it further without the actual device.

Hi Obmaz, i also have this device. Can I help you with debugging to make the knob working? I also have a tuya gateway so i can execute something in smartlife if this helps.

obmaz, 2025-02-11:
Working without the actual device is practically impossible. If you provide me with a link to the same product on AliExpress, I’ll purchase it and run some tests.

marcos-mvs, 2025-02-11:
Thanks Obmaz! Mine is the same model, from this link: https://a.aliexpress.com/_mMU5IdV
Tomorrow I’ll try to record some logs in the CLI

marcos-mvs, 2025-02-11:
[tuyaknob.txt](https://github.com/user-attachments/files/18754572/tuyaknob.txt)

obmaz, 2025-02-19:
@Staffed24 @marcos-mvs 

The device I purchased from AliExpress has arrived, and I updated the driver slightly.
The UI on the detail screen has also changed. (Now, only one button appears, but there are five actions.)
The device must be deleted and reconnected for the changes to take effect.

This device supports two modes, and the features supported by this driver are as follows:
Pressing the button three times quickly switches the mode.

Remote Mode:
	•	Push: Working
	•	Double Push: Working
	•	Long Push (Release): Working
	•	Knob: Working, but it only functions as an On/Off switch rather than adjusting values based on rotation.

Scene Mode:
	•	Push: Working
	•	Double Push: Device does not support in device
	•	Long Push: Device does not support in device
	•	Knob: Not working

Technical Details:

In Scene Mode, the knob operation sends a value through the "Step" ZCL cluster, which does not seem to be part of the Zigbee standard. As a result, the SmartThings ZCL library does not parse it correctly.

To support this, the driver would need to be implemented without using st.zigbee.defaults.register_for_default_handlers. However, this is a highly complex task, making it difficult for an individual to handle alone. Other developers are likely facing a similar challenge.

Lastly, if you tag zambobmaz on the SmartThings Community, I can check it via email.
I read all posts and comments with the tag—I just don’t actively participate in the community. 🙂

(Driver Version 2025-02-19T15:21:38.643732774)

marcos-mvs, 2025-02-19:
@obmaz , thanks for your support and splendid work!

It seems to me that the scene mode is to be used when grouped/binded directly to a supported device through touchlink.
That said, with "remote mode" working, I believe I can use Mariano's driver to increase or decrease the intensity through the "Level by External Steps" function. That's what I'm trying to achieve.

Installed and made some test with the new driver version, here is some observations:
- The Device Details screen correctly shows one button, but only three actions: "pushed","double","held". The same is true for the actions that are available in routines:

[image]
[image]
[image]
<br><br>

- Although, it's possible to see in the Device Details and History screen that the Down/Up event are being fired:

[image]
[image]
[image]

- It's possible to enable rotate Up/Down actions on the Routines?
- The device only reports the state of rotate Up/Down when the rotations stops, right?, So we can't fire the Up/Down action at every click. That would be interesting, or maybe it could report the number of clicks, but it's acceptable. I don't remember if on the Tuya hub it had this same behavior, I can check.
- For the "Held" action, the device only reports the state when the button is released? This is a little odd, because if you hold the button for 2 or 3 seconds, it doesn´t fire any action. You have to headcount 4 or 5 seconds and then release to get the "Held" action.
- Also, this device seems to have two more actions: push (Held) and rotate Up OR Down. Are these actions correctly sent to the hub in remote mode? Perhaps this is only an option in scene mode!

obmaz, 2025-02-20:
@marcos-mvs 

Answer

> It's possible to enable rotate Up/Down actions on the Routines?

- Yes, I implemented

>Edit: If you continue to rotate, the knob seems to report one event every aprox. 2s. So, we can't fire the Up/Down action at every click, but it's acceptable. I don't remember if on the Tuya hub it had this same behavior, I can check.

- Since this signal comes from the device, it cannot be processed.

>For the "Held" action, the device only reports the state when the button is released? This is a little odd, because if you hold the button for 2 or 3 seconds, it doesn´t fire any action. You have to headcount 4 or 5 seconds and then release to get the "Held" action.

- The signal is received upon release. You can check it in the log.

>Also, this device seems to have two more actions: push (Held) and rotate Up OR Down. Are these actions correctly sent to the hub in remote mode? Perhaps this is only an option in scene mode!

- Yes, I implemented

Through several tests today, I have almost completed the implementation.
When the knob is directly connected to the device, brightness changes (like with a bulb) are smooth. However, when passing through the hub, there are some limitations.

I have created a new profile. As always, you need to delete and reconnect the device.
If the screen does not appear as shown in the attached image, please try again the next day. It seems to take some time for deployment.

Remote Mode:
• Push: Working
• Double Push: Working
• Long Push (Release): Working
• Knob: Working
• Knob Level: Device does not support
• Knob with hold: Device does not support

[image]

Scene Mode:
• Push: Working
• Double Push: Device does not support 
• Long Push: Device does not support
• Knob: Working
• Knob Level: Working
• Knob with hold: Working
• Knob Level  with hold: Working

I have added wheel and hold_wheel level. These values represent the angle the knob is turned.
• Turning clockwise increases the value.
• Turning counterclockwise decreases the value.
• The reference point is 50.
• This value is generated each time the knob is turned and is not accumulated from previous values.

FYI:
version: 2025-02-20T16:13:20.739135987
support manufacture: _TZ3000_402vrq2i
support model: TS004F

[image]

marcos-mvs, 2025-03-11:
Hi @obmaz , sorry for the delay in response. 
Been doing some tests with the driver and have some observations and issues.

First, but not more important, i think that in the manual nomenclature, remote mode is when the knob is bonded directly to other devices and scene is when it fires events to the hub.
It’s particularly odd because the scene mode it’s supposed to be the default, but whenever i delete and reconnect the device, it always start in remote mode.

That said, when in scene mode, that you call remote, the device works as expected but wheel and hold_wheel level never gets initialized:

[image]

When in remote mode, that you call scene, nothing work in the app, no events are fired, neither the wheel levels.

How do you bind the device directly to the bulb to control?

obmaz, 2025-03-11:
@marcos-mvs 

There are a few things that need to be clarified first.

I don’t know how to directly connect the Tuya Knob to a bulb. What I mentioned earlier was about general Zigbee direct binding. Some switches sold by IKEA work this way.

1. The mode set when the device is initially paired with the SmartThings hub cannot be configured via the driver.The only way to change it is by manually pressing the button three times quickly.

2. I don’t understand what you mean by:	
- “when in scene mode, that you call remote”
- “When in remote mode, that you call scene”

The driver I created processes all incoming signals from the knob, regardless of whether it is in remote mode or scene mode. In other words, the driver functions regardless of the mode.

The only difference, as described above, is that the actual data sent by the device differs between remote mode and scene mode, and the driver operates accordingly.

For example, when turning the knob in remote mode, no specific value is transmitted—only the fact that the knob has been turned is detected.

However, in scene mode, turning the knob transmits an actual value indicating how much it was turned.
When the knob is turned in scene mode, the turned value is displayed, and it does not automatically reset to 50.
It is not possible to determine when it should be reset, and if it were to be automatically reset to 50, unintended automation could occur.

marcos-mvs, 2025-03-12:
OK, that clarifies a lot of things :)

Deleted em repaired the device several times. Remote Mode is working fine, and all these events are correctly triggered:
• Push: Working
• Double Push: Working
• Long Push (Release): Working
• Knob: Working

But when pressing the button three times quickly to change to Scene mode, the device stops responding. No more events triggered. I can't reproduce this behaviour:

Scene Mode:
• Push: Working
• Double Push: Device does not support
• Long Push: Device does not support
• Knob: Working
• Knob Level: Working
• Knob with hold: Working
• Knob Level with hold: Working

obmaz, 2025-03-12:
Scene Mode log looks like empty. only show "zigbee-tuya-button  driver device thread event handled"

I need a log that is similar to the following one.

Note:
Remote Mode : ZCLCommandId: 0xFC / 0xFD
Scene Mode : ZCLCommandId: 0x02

<ZigbeeDevice: 74ee34cc-5c5d-49d3-b4a8-a39f5fa2a13a [0x6423] (Tuya Knob)> received Zigbee message: < ZigbeeMessageRx || type: 0x01, < AddressHeader || src_addr: 0x4408, src_endpoint: 0x01, dest_addr: 0x0000, dest_endpoint: 0xFF, profile: 0x0104, cluster: OnOff >, lqi: 0xFF, rssi: -61, body_length: 0x0003, < ZCLMessageBody || < ZCLHeader || frame_ctrl: 0x01, seqno: 0x26, ZCLCommandId: 0x02 >, < Toggle ||  > > >

<ZigbeeDevice: 74ee34cc-5c5d-49d3-b4a8-a39f5fa2a13a [0x6423] (Tuya Knob)> received Zigbee message: < ZigbeeMessageRx || type: 0x01, < AddressHeader || src_addr: 0x6423, src_endpoint: 0x01, dest_addr: 0x0000, dest_endpoint: 0xFF, profile: 0x0104, cluster: Level >, lqi: 0xFF, rssi: -58, body_length: 0x0007, < ZCLMessageBody || < ZCLHeader || frame_ctrl: 0x01, seqno: 0xCE, ZCLCommandId: 0x02 >, < Step || MoveStepMode: UP, step_size: 0x49, transition_time: 0x0002 > > >

obmaz, 2025-03-12:
It seems that the process might have terminated before the log was transmitted.
In any case, if there is a log in scene mode, please send it again.
As mentioned above, the important part is the section where zclCommandId appears.

obmaz, 2025-03-12:
That’s strange.
I’m not sure why, but it seems that no signal is coming in scene mode.
It could be a hardware issue

marcos-mvs, 2025-03-12:
> That’s strange. I’m not sure why, but it seems that no signal is coming in scene mode. It could be a hardware issue

Don’t think it’s a hardware issue, cause i added the knob to the Tuya Hub with Smart Life app. In Smart Life, the app interface changes automatically when switching between remote and scene mode. In Scene mode, it let me choose another devices to form a “group” . I tested grouping with a bulb and the bulb respond accordingly: on/off on button presses and dimmer level with rotate actions.

marcos-mvs, 2025-03-21:
@obmaz , doing some aditional tests with the knob
While in the "remote mode" i noticed some odd messages in the logs, but, again, nothing comes in when in scene mode.
Full log: [20250321 knob log 1.txt](https://github.com/user-attachments/files/19396609/20250321.knob.log.1.txt)

for example:
2025-03-21T19:37:14.154229327Z TRACE zigbee-tuya-button  Received event with handler device_lifecycle
2025-03-21T19:37:14.154362244Z INFO zigbee-tuya-button  <ZigbeeDevice: db1f191a-2d7a-45ac-b9a6-4b5ea3d9ab0c [0x9125] (Tuya Knob)> received lifecycle event: infoChanged
2025-03-21T19:37:14.158155327Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 :       <ZigbeeDevice: db1f191a-2d7a-45ac-b9a6-4b5ea3d9ab0c [0x9125] (Tuya Knob)>
2025-03-21T19:37:14.158285994Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 :       <ZigbeeDevice: db1f191a-2d7a-45ac-b9a6-4b5ea3d9ab0c [0x9125] (Tuya Knob)>
2025-03-21T19:37:14.158867327Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 : false
2025-03-21T19:37:14.158999869Z TRACE zigbee-tuya-button  <ZigbeeDevice: db1f191a-2d7a-45ac-b9a6-4b5ea3d9ab0c **[0x9125] (Tuya Knob)> received unhandled lifecycle event: infoChanged**
2025-03-21T19:37:14.159526119Z DEBUG zigbee-tuya-button  Tuya Knob device thread event handled

2025-03-21T19:40:39.965912239Z TRACE zigbee-tuya-button  Received event with handler zigbee
2025-03-21T19:40:39.966049364Z INFO zigbee-tuya-button  <ZigbeeDevice: db1f191a-2d7a-45ac-b9a6-4b5ea3d9ab0c [0x9125] (Tuya Knob)> received Zigbee message: < ZigbeeMessageRx || type: 0x00, < AddressHeader || src_addr: 0x9125, src_endpoint: 0x01, dest_addr: 0x0000, dest_endpoint: 0x01, profile: 0x0104, cluster: Groups >, lqi: 0xFF, rssi: -55, body_length: 0x0005, < ZCLMessageBody || < ZCLHeader || frame_ctrl: 0x19, seqno: 0x06, ZCLCommandId: 0x02 >, < GetGroupMembershipResponse || capacity: 0x10, group_list_length: 0x00 > > >
2025-03-21T19:40:39.969407906Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 :       <ZigbeeDevice: db1f191a-2d7a-45ac-b9a6-4b5ea3d9ab0c [0x9125] (Tuya Knob)>
2025-03-21T19:40:39.969547698Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 :       <ZigbeeDevice: db1f191a-2d7a-45ac-b9a6-4b5ea3d9ab0c [0x9125] (Tuya Knob)>
2025-03-21T19:40:39.970263239Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 : false
2025-03-21T19:40:39.970402114Z DEBUG zigbee-tuya-button  Tuya Knob device thread event handled

2025-03-21T19:40:39.978022906Z TRACE zigbee-tuya-button  Received event with handler zigbee
2025-03-21T19:40:39.978157573Z INFO zigbee-tuya-button  <ZigbeeDevice: db1f191a-2d7a-45ac-b9a6-4b5ea3d9ab0c [0x9125] (Tuya Knob)> received Zigbee message: < ZigbeeMessageRx || type: 0x00, < AddressHeader || src_addr: 0x9125, src_endpoint: 0x01, dest_addr: 0x0000, dest_endpoint: 0x01, profile: 0x0104, cluster: Groups >, lqi: 0xFF, rssi: -55, body_length: 0x0005, < ZCLMessageBody || < ZCLHeader || frame_ctrl: 0x18, seqno: 0x06, ZCLCommandId: 0x0B >, < DefaultResponse || cmd: 0x02, ZclStatus: SUCCESS > > >
2025-03-21T19:40:39.982185489Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 :       <ZigbeeDevice: db1f191a-2d7a-45ac-b9a6-4b5ea3d9ab0c [0x9125] (Tuya Knob)>
2025-03-21T19:40:39.983087031Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 :       <ZigbeeDevice: db1f191a-2d7a-45ac-b9a6-4b5ea3d9ab0c [0x9125] (Tuya Knob)>
2025-03-21T19:40:39.983936406Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 : false
2025-03-21T19:40:39.984801823Z DEBUG zigbee-tuya-button  Tuya Knob device thread event handled