obmaz/smartthings_edge_driver #55: Tuya Zigbee Button Stopped Working Correctly

marcos-mvs, 2025-03-14:
Device fingerprint: _TZ3000_ja5osu5g

I have a Tuya wireless button set up with smart things to toggle on/off a smart bulb. Button sems to be sending multiple press events, because sometimes the lamp turns off and then turns back on again.
Wonder if the button fires a "toggle up"event on button release and the drivers handle as another single press

This is the log collected when single pressing the button: 

> 2025-03-14T00:02:27.606994966Z TRACE zigbee-tuya-button  Received event with handler zigbee
2025-03-14T00:02:27.607713466Z INFO zigbee-tuya-button  <ZigbeeDevice: 1424725f-fc75-44a8-aaec-f932e9f2ee0a [0x07C6] (Controle Luminária Mesa)> received Zigbee message: < ZigbeeMessageRx || type: 0x00, < AddressHeader || src_addr: 0x07C6, src_endpoint: 0x01, dest_addr: 0x0000, dest_endpoint: 0x01, profile: 0x0104, cluster: OnOff >, lqi: 0xFF, rssi: -44, body_length: 0x0004, < ZCLMessageBody || < ZCLHeader || frame_ctrl: 0x11, seqno: 0x5F, ZCLCommandId: 0xFD >, GenericBody:  00 > >
2025-03-14T00:02:27.613137424Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 :       <ZigbeeDevice: 1424725f-fc75-44a8-aaec-f932e9f2ee0a [0x07C6] (Controle Luminária Mesa)>
2025-03-14T00:02:27.614941382Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 :       <ZigbeeDevice: 1424725f-fc75-44a8-aaec-f932e9f2ee0a [0x07C6] (Controle Luminária Mesa)>
2025-03-14T00:02:27.616750132Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 : false
2025-03-14T00:02:27.618795132Z TRACE zigbee-tuya-button  Found ZigbeeMessageDispatcher handler in zigbee-tuya-button
2025-03-14T00:02:27.620438382Z INFO zigbee-tuya-button  Executing ZclClusterCommandHandler: cluster: OnOff, command: 0xFD
2025-03-14T00:02:27.623951591Z INFO zigbee-tuya-button  <<---- Moon ---->> button_handler       1
2025-03-14T00:02:27.624526341Z INFO zigbee-tuya-button  <ZigbeeDevice: 1424725f-fc75-44a8-aaec-f932e9f2ee0a [0x07C6] (Controle Luminária Mesa)> emitting event: {"attribute_id":"button","capability_id":"button","component_id":"button1","state":{"value":"pushed"},"state_change":true}
2025-03-14T00:02:27.667190924Z DEBUG zigbee-tuya-button  Controle Luminária Mesa device thread event handled
2025-03-14T00:02:27.864585757Z TRACE zigbee-tuya-button  Received event with handler zigbee
2025-03-14T00:02:27.865168507Z INFO zigbee-tuya-button  <ZigbeeDevice: 1424725f-fc75-44a8-aaec-f932e9f2ee0a [0x07C6] (Controle Luminária Mesa)> received Zigbee message: < ZigbeeMessageRx || type: 0x00, < AddressHeader || src_addr: 0x07C6, src_endpoint: 0x01, dest_addr: 0x0000, dest_endpoint: 0x01, profile: 0x0104, cluster: OnOff >, lqi: 0xFF, rssi: -44, body_length: 0x0003, < ZCLMessageBody || < ZCLHeader || frame_ctrl: 0x11, seqno: 0x60, ZCLCommandId: 0x02 >, < Toggle ||  > > >
2025-03-14T00:02:27.874773049Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 :       <ZigbeeDevice: 1424725f-fc75-44a8-aaec-f932e9f2ee0a [0x07C6] (Controle Luminária Mesa)>
2025-03-14T00:02:27.875418757Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 :       <ZigbeeDevice: 1424725f-fc75-44a8-aaec-f932e9f2ee0a [0x07C6] (Controle Luminária Mesa)>
2025-03-14T00:02:27.876239716Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 : false
2025-03-14T00:02:27.876846007Z TRACE zigbee-tuya-button  Found ZigbeeMessageDispatcher handler in zigbee-tuya-button
2025-03-14T00:02:27.877700924Z INFO zigbee-tuya-button  Executing ZclClusterCommandHandler: cluster: OnOff, command: Toggle
2025-03-14T00:02:27.878651674Z INFO zigbee-tuya-button  <<---- Moon ---->> button_handler_Knob_Scene_Toggle     1
2025-03-14T00:02:27.879204882Z INFO zigbee-tuya-button  <ZigbeeDevice: 1424725f-fc75-44a8-aaec-f932e9f2ee0a [0x07C6] (Controle Luminária Mesa)> emitting event: {"attribute_id":"button","capability_id":"button","component_id":"button1","state":{"value":"pushed"},"state_change":true}
2025-03-14T00:02:27.919929799Z DEBUG zigbee-tuya-button  Controle Luminária Mesa device thread event handled

marcos-mvs, 2025-03-14:
this is a double press log:
> 2025-03-14T00:19:24.175437397Z INFO zigbee-tuya-button  <ZigbeeDevice: 1424725f-fc75-44a8-aaec-f932e9f2ee0a [0x07C6] (Controle Luminária Mesa)> received Zigbee message: < ZigbeeMessageRx || type: 0x00, < AddressHeader || src_addr: 0x07C6, src_endpoint: 0x01, dest_addr: 0x0000, dest_endpoint: 0x01, profile: 0x0104, cluster: OnOff >, lqi: 0xFF, rssi: -46, body_length: 0x0004, < ZCLMessageBody || < ZCLHeader || frame_ctrl: 0x11, seqno: 0x6F, ZCLCommandId: 0xFD >, GenericBody:  01 > >
2025-03-14T00:19:24.184642647Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 :       <ZigbeeDevice: 1424725f-fc75-44a8-aaec-f932e9f2ee0a [0x07C6] (Controle Luminária Mesa)>
2025-03-14T00:19:24.185274397Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 :       <ZigbeeDevice: 1424725f-fc75-44a8-aaec-f932e9f2ee0a [0x07C6] (Controle Luminária Mesa)>
2025-03-14T00:19:24.185893813Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 : false
2025-03-14T00:19:24.186718522Z TRACE zigbee-tuya-button  Found ZigbeeMessageDispatcher handler in zigbee-tuya-button
2025-03-14T00:19:24.187343605Z INFO zigbee-tuya-button  Executing ZclClusterCommandHandler: cluster: OnOff, command: 0xFD
2025-03-14T00:19:24.188504813Z INFO zigbee-tuya-button  <<---- Moon ---->> button_handler       1
2025-03-14T00:19:24.189068522Z INFO zigbee-tuya-button  <ZigbeeDevice: 1424725f-fc75-44a8-aaec-f932e9f2ee0a [0x07C6] (Controle Luminária Mesa)> emitting event: {"attribute_id":"button","capability_id":"button","component_id":"button1","state":{"value":"double"},"state_change":true}
2025-03-14T00:19:24.210910730Z DEBUG zigbee-tuya-button  Controle Luminária Mesa device thread event handled
2025-03-14T00:19:24.458152480Z TRACE zigbee-tuya-button  Received event with handler zigbee
2025-03-14T00:19:24.526772813Z INFO zigbee-tuya-button  <ZigbeeDevice: 1424725f-fc75-44a8-aaec-f932e9f2ee0a [0x07C6] (Controle Luminária Mesa)> received Zigbee message: < ZigbeeMessageRx || type: 0x00, < AddressHeader || src_addr: 0x07C6, src_endpoint: 0x01, dest_addr: 0x0000, dest_endpoint: 0x01, profile: 0x0104, cluster: OnOff >, lqi: 0xFF, rssi: -46, body_length: 0x0003, < ZCLMessageBody || < ZCLHeader || frame_ctrl: 0x11, seqno: 0x70, ZCLCommandId: 0x01 >, < On ||  > > >
2025-03-14T00:19:24.586170063Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 :       <ZigbeeDevice: 1424725f-fc75-44a8-aaec-f932e9f2ee0a [0x07C6] (Controle Luminária Mesa)>
2025-03-14T00:19:24.606793772Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 :       <ZigbeeDevice: 1424725f-fc75-44a8-aaec-f932e9f2ee0a [0x07C6] (Controle Luminária Mesa)>
2025-03-14T00:19:24.616809855Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 : false
2025-03-14T00:19:24.619264022Z DEBUG zigbee-tuya-button  Controle Luminária Mesa device thread event handled

marcos-mvs, 2025-03-14:
This is a HOLD log:
> 2025-03-14T00:21:44.801337562Z TRACE zigbee-tuya-button  Received event with handler zigbee
2025-03-14T00:21:44.806678437Z INFO zigbee-tuya-button  <ZigbeeDevice: 1424725f-fc75-44a8-aaec-f932e9f2ee0a [0x07C6] (Controle Luminária Mesa)> received Zigbee message: < ZigbeeMessageRx || type: 0x00, < AddressHeader || src_addr: 0x07C6, src_endpoint: 0x01, dest_addr: 0x0000, dest_endpoint: 0x01, profile: 0x0104, cluster: OnOff >, lqi: 0xFF, rssi: -49, body_length: 0x0004, < ZCLMessageBody || < ZCLHeader || frame_ctrl: 0x11, seqno: 0x71, ZCLCommandId: 0xFD >, GenericBody:  02 > >
2025-03-14T00:21:44.847250103Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 :       <ZigbeeDevice: 1424725f-fc75-44a8-aaec-f932e9f2ee0a [0x07C6] (Controle Luminária Mesa)>
2025-03-14T00:21:44.856997270Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 :       <ZigbeeDevice: 1424725f-fc75-44a8-aaec-f932e9f2ee0a [0x07C6] (Controle Luminária Mesa)>
2025-03-14T00:21:44.858014478Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 : false
2025-03-14T00:21:44.858809437Z TRACE zigbee-tuya-button  Found ZigbeeMessageDispatcher handler in zigbee-tuya-button
2025-03-14T00:21:44.859360603Z INFO zigbee-tuya-button  Executing ZclClusterCommandHandler: cluster: OnOff, command: 0xFD
2025-03-14T00:21:44.878212978Z INFO zigbee-tuya-button  <<---- Moon ---->> button_handler       1
2025-03-14T00:21:44.894431353Z INFO zigbee-tuya-button  <ZigbeeDevice: 1424725f-fc75-44a8-aaec-f932e9f2ee0a [0x07C6] (Controle Luminária Mesa)> emitting event: {"attribute_id":"button","capability_id":"button","component_id":"button1","state":{"value":"held"},"state_change":true}
2025-03-14T00:21:45.001348145Z DEBUG zigbee-tuya-button  Controle Luminária Mesa device thread event handled
2025-03-14T00:21:45.007852645Z TRACE zigbee-tuya-button  Received event with handler zigbee
2025-03-14T00:21:45.014746103Z INFO zigbee-tuya-button  <ZigbeeDevice: 1424725f-fc75-44a8-aaec-f932e9f2ee0a [0x07C6] (Controle Luminária Mesa)> received Zigbee message: < ZigbeeMessageRx || type: 0x00, < AddressHeader || src_addr: 0x07C6, src_endpoint: 0x01, dest_addr: 0x0000, dest_endpoint: 0x01, profile: 0x0104, cluster: OnOff >, lqi: 0xFF, rssi: -48, body_length: 0x0003, < ZCLMessageBody || < ZCLHeader || frame_ctrl: 0x11, seqno: 0x72, ZCLCommandId: 0x00 >, < Off ||  > > >
2025-03-14T00:21:45.030059228Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 :       <ZigbeeDevice: 1424725f-fc75-44a8-aaec-f932e9f2ee0a [0x07C6] (Controle Luminária Mesa)>
2025-03-14T00:21:45.030700937Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 :       <ZigbeeDevice: 1424725f-fc75-44a8-aaec-f932e9f2ee0a [0x07C6] (Controle Luminária Mesa)>
2025-03-14T00:21:45.031412978Z INFO zigbee-tuya-button  <<---- Moon ---->> is_tuya_ef00 : false
2025-03-14T00:21:45.031949020Z DEBUG zigbee-tuya-button  Controle Luminária Mesa device thread event handled

obmaz, 2025-03-14:
It seems that the TS004F 1-button device sends both remote mode and scene mode signals simultaneously.
When adding the knob function, I configured it to handle both signals, which appears to be causing the malfunction.
I will fix it soon.

marcos-mvs, 2025-03-14:
> [@marcos-mvs](https://github.com/marcos-mvs) I fixed it and depolyed (2025-03-14T11:30:03.143778274) Could you check whether works?

working great again! Thanks

closed 2025-03-14 by marcos-mvs